schmurtzm/MrDiy-Audio-Notifier #5: mrdiy

alfovit, 2022-02-18:
what is the username and password to login to 192.168.4.1      ????

schmurtzm, 2022-02-18:
The login is "admin" and the password is the "AP password" that you have defined during wifi configuration.
It's not intuitive, I will specify it in the readme.
[image]

alfovit, 2022-02-18:
I can't set up wifi because I can't access the address 192.168.4.1, it asks for a username and password

schmurtzm, 2022-02-18:
On the first connection it should not ask for password.
Erase the flash of your ESP and then reflash it.  You can use the script "Schmurtz_ESP_Flasher.bat" to dot it.

schmurtzm, 2022-02-19:
I have added the informations in the readme now.
Is it OK for you after erasing the flash ?

alfovit, 2022-02-20:
re-flashed, everything worked out, thanks

schmurtzm, 2022-02-20:
OK great, so I close this one ;)

closed 2022-02-20 by schmurtzm