schmurtzm/MrDiy-Audio-Notifier #6: [BUG] - Stop command doesn't work

schmurtzm, 2022-02-20:
When playing a long mp3 (like a web radio) MQTT commands are not received anymore.
So it is impossible to stop, play something else, set volume... 
Also LWT MQTT status goes offline.

Not sure that it is possible to continue listen MQTT during playing with ESP8266audio library....

Hurstolds, 2022-02-23:
I tested this with local hosted MP3 files and stop command worked.    I haven't tested with a web radio.  I recommend closing this issue out if that works for you!

xanderphillips, 2023-01-12:
Just found this project tonight, very nice!
I just tested with the example icecast stream against my ESP32 and it does exactly what was stated in this bug, which is what led me to search the Issues on github.   If I have the stream playing I can no longer issue MQTT commands like stop or change volume.  The web interface also appears to become unresponsive.  The only way I can stop the stream and regain control is to reboot the ESP32.