schmurtzm/MrDiy-Audio-Notifier #7: Google Voice crashes

Hurstolds, 2022-02-21:
Hello,

I had this issue in 0.4 and wanted to test out 0.5 before I posted this.  When I send a /googlevoice/ broadcast, it works once.  After speaking the text, the unit essentially crashes.  MQTT status does show offline after a short while.  The only work around I have found is to power cycle the unit.

I saw your other bug post, and they may be related.  I don't do much with MP3 streaming yet, so I haven't tested that against the new 0.5 code.

The /samvoice/ command doesn't exhibit the same issue as the /googlevoice/ command.  Sam voice works normal.  Once it's done speaking the text, MQTT reports back that the speaker is idle.  I'm only seeing issues with /googlevoice/ command.

schmurtzm, 2022-02-21:
Hi,

Yes, this is an identified bug related to esp8266audio library.
I've created an issue in the esp8266audio repo  : https://github.com/earlephilhower/ESP8266Audio/issues/475   (it was also identifier before in [this issue](https://github.com/earlephilhower/ESP8266Audio/issues/395))

I can't manage to solve this by myself, so we have to wait for a coding hero 😄 
However I've explained the origine of the problem [here](https://github.com/earlephilhower/ESP8266Audio/issues/475#issuecomment-1011550500). In fact the ESP doesn't crash : it just don't know that the mp3 is finished so he think he is playing something during few minutes then it stop playing and works again normally. It is related to chunk management in ESP8266audio. Someone has already detected this lack of chunk management and has made [a Pull Request,](https://github.com/earlephilhower/ESP8266Audio/pull/394) when You use his code the ESP hang only few seconds (instead of minutes) after playing Google TTS. So it's not perfect but still a good improvement ;)
EDIT : I've just seen that [the PR](https://github.com/earlephilhower/ESP8266Audio/pull/394) has been updated some guys talk about GoogleTTS... So there's a little chance, I will test it.

I've already tested another audio library which works very fine with Google TTS but it has drawbacks : less functionalities and compatible only with esp32. May be I will create a new MQTT audio Notifier in the future...

However if you use Home Assistant there are some workarounds which work great. 
The best solution is to transform MrDIY Notifier into a media player (which supports TTS). 
Because when you use tts in HA it generate an MP3, put it on his web server and then ask a player to play it. Thanks to [this post](https://community.home-assistant.io/t/media-player-template/203062/91) you will be able to transform MrDIY Notifier into Media Player. It require first to install Media_player.template on your HA (easy thanks to hacs) and then add my configuration in your configuration.yaml. On Home Assistant Forum I also talk about other [alternative solutions](https://community.home-assistant.io/t/turn-an-esp8266-wemosd1mini-into-an-audio-notifier-for-home-assistant-play-mp3-tts-rttl/211499/128?u=schmurtz) which definitely deserve a try ;)
I remind that MrDIY is an audio Notifier, if you use it like a real media player you will be probably disappointed due to the lack of compatibility with different audio formats. So if you look to an awesome media player for ESP32 take a look to [squeezelite-esp32](https://github.com/sle118/squeezelite-esp32) This project is really incredible !

schmurtzm, 2022-02-21:
OK google TTS seems to works thanks to [DeqingSun post](https://github.com/earlephilhower/ESP8266Audio/pull/394#issuecomment-1030737629)  🙌 
Let's make a new release ;)

schmurtzm, 2022-02-22:
Version 0.6 released, it's not perfect but a lot better than before !

Hurstolds, 2022-02-23:
All seems well, after testing 0.61!  Thank you, I did not have to do a full flash.  I was able to flash via GUI updater and it kept all my previous settings.  Great job and thank you again!

closed 2022-02-23 by Hurstolds