schmurtzm/MrDiy-Audio-Notifier #9: Version 0.6 doesn't support multiple notifiers

Hurstolds, 2022-02-22:
After some further testing, here is what I’m seeing. I am running two of these speaker notifiers.
Speaker 1 was running firmware 0.4
Speaker 2 was running firmware 0.5

I upgraded speaker 2 to version 0.6 and it works, just samvoice is broken.

I just now upgraded speaker 1 to version 0.6 and there’s a problem running both at version 0.6 This was an issue I saw when running Mr DIY original firmware one more than one speaker. It’s basically created a MQTT storm, where both units keep firing off messages saying they’re idle and connected. When this happens, I can’t send commands to either speaker.

To further test, I downgraded speaker 1 to version 0.5. speaker 2 running version 0.6 and it has the same problem. Both are firing messages non stop via MQTT, causing neither of them to work.

So I stepped speaker 1 back down to version 0.4 . Speaker 2 is running latest 0.6 version, and now both work as expected. I presume version 0.5 introduced an issue that causes multiple notifiers to break.

One further test to confirm:

If both speaker/notifiers are running firmware version 0.4, they both work correctly. No issues with contention and no MQTT storm of broadcasts from the notifiers.

schmurtzm, 2022-02-22:
Yes I see what happens here. You can solve it by changing the "AP SSID" name with different name for your 2 devices..

Before the name was different for each device, for simplicity I put an unique name (it allow to access to your device with "http://mrdiynotifier.local/") but it was a bad idea.

This "AP SSID" field in realty is used for 3 different things : 
- The AP name for configuration when you're not connected to your own wifi
- The DNS name to access to your web configuration UI with the url "http://mrdiynotifier.local/"
- The MQTT client ID to use when connecting to the server 

I will revert back this modification on next version.

schmurtzm, 2022-02-22:
0.61 released, it should support multiple notifiers again. May be you will need to erase flash to generate new names.

Hurstolds, 2022-02-23:
All seems well, after testing 0.61!  Thank you, I did not have to do a full flash.  I was able to flash via GUI updater and it kept all my previous settings.  Great job and thank you again!

closed 2022-02-23 by Hurstolds