schmurtzm/MrDiy-Audio-Notifier #11: Help required

ChrisSutton123, 2022-03-16:
Unable to play any sound
I uploaded the precompiled "FW_v0.61_ESP8266_d1_mini.bin" to the board
I am using a Wemos  D1 mini, configured for NODAC.
I have done initial setup, D1 connects successfully. System plays the boot_sound.h file okay
However I cannot get anything to play I have tried "play", "samvoice" and "googlevoice" no sound is output

Initially I was using "MrDIY_notifier_custom_MQTT.bin"  without any problem.

I can toggle between the firmware's using the same MQTT topic's one works the other doesn't .

Ultimately I will not be using the "NODAC" option I am waiting for delivery of Adafruit I2S 3W Class D Amplifier Breakout - MAX98357A. For now I would like to get the "NODAC" option working.

Is there any debug output I can check?

certeza, 2022-03-17:
I noticed that too. Maybe the buffer is emptied too fast  before the next call.

ChrisSutton123, 2022-03-18:
If its any help the s/w versions I am using are:

ESP8266          - 3.0.2
ESP8266Audio - 1.9.5
ESP8266SAM   - 1.0.1

Also in the readme for ESP8266Audio it mentions to use:-

Tools->lwIP Variant->v1.4 Open Source, or V2 Higher Bandwidth
Tools->CPU Frequency->160MHz

I tried that didn't fix my problem

I would like to be able to build the binary from your source and get the same 
sound output result as your supplied prebuilt binary.
 
What additional setting am I missing?

ChrisSutton123, 2022-03-18:
Sorry again. Not your problem!
I reinstalled my Visual Studio Code development environment.
Now my built binary performs the same as you binary.

closed 2022-03-18 by ChrisSutton123