schmurtzm/MrDiy-Audio-Notifier #20: Not compile

nocturno66, 2022-07-22:
Hi Schmurtzm,
I have a problem compiling your code with library IotWebConf. I have installed versión 3.2.0 of these library but the error persist.

You can view the LOG attached.

Can you help me?
Thanks you in advance

Processing nodemcuv2 (platform: espressif8266; framework: arduino; board: esp12e)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (3.2.0) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)       
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
Converting esp8266_mrdiy_mqtt_local_notifier.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP8266Audio @ 1.9.7
|   |-- I2S @ 1.0
|   |-- ESP8266HTTPClient @ 1.2
|   |   |-- ESP8266WiFi @ 1.0
|   |-- SD @ 2.0.0
|   |   |-- SDFS @ 0.1.0
|   |   |   |-- ESP8266SdFat @ 2.0.2
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|-- ESP8266SAM @ 1.0.1
|   |-- ESP8266Audio @ 1.9.7
|   |   |-- I2S @ 1.0
|   |   |-- ESP8266HTTPClient @ 1.2
|   |   |   |-- ESP8266WiFi @ 1.0
|   |   |-- SD @ 2.0.0
|   |   |   |-- SDFS @ 0.1.0
|   |   |   |   |-- ESP8266SdFat @ 2.0.2
|   |   |   |   |   |-- SPI @ 1.0
|   |   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|-- PubSubClient @ 2.8.0
|-- IotWebConf @ 3.2.0
|   |-- DNSServer @ 1.1.1
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WebServer @ 1.0
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- EEPROM @ 1.0
|   |-- ESP8266mDNS @ 1.2
|   |   |-- ESP8266WiFi @ 1.0
|-- ESP8266WiFi @ 1.0
Building in release mode
Compiling .pio\build\nodemcuv2\src\esp8266_mrdiy_mqtt_local_notifier.ino.cpp.o
Generating LD script .pio\build\nodemcuv2\ld\local.eagle.app.v6.common.ld
Compiling .pio\build\nodemcuv2\libd5d\I2S\I2S.cpp.o
Archiving .pio\build\nodemcuv2\libd5d\libI2S.a
Compiling .pio\build\nodemcuv2\lib879\ESP8266WiFi\BearSSLHelpers.cpp.o
**C:/Users/nocturno/Documents/PlatformIO/Projects/220721-131454-nodemcuv2/src/esp8266_mrdiy_mqtt_local_notifier.ino:139:1: error: 'IotWebConfParameter' does not name a type; did you mean 'IotWebConfParameter_h'?
  139 | IotWebConfParameter   mqttServerParam       = IotWebConfParameter("MQTT server", "mqttServer", mqttServer, sizeof(mqttServer) );**
      | ^~~~~~~~~~~~~~~~~~~
      | IotWebConfParameter_h
C:/Users/nocturno/Documents/PlatformIO/Projects/220721-131454-nodemcuv2/src/esp8266_mrdiy_mqtt_local_notifier.ino:140:1: error: 'IotWebConfParameter' does not name a type; did you mean 'IotWebConfParameter_h'?
  140 | IotWebConfParameter   mqttUserNameParam     = IotWebConfParameter("MQTT username", "mqttUser", mqttUserName, sizeof(mqttUserName));
      | ^~~~~~~~~~~~~~~~~~~
      | IotWebConfParameter_h
[... 46 lines cut ...]
.pio\libdeps\nodemcuv2\IotWebConf\src/IotWebConf.h:579:18: note: declared private here
  579 |   ParameterGroup _allParameters = ParameterGroup("iwcAll");
      |                  ^~~~~~~~~~~~~~
C:/Users/nocturno/Documents/PlatformIO/Projects/220721-131454-nodemcuv2/src/esp8266_mrdiy_mqtt_local_notifier.ino:162:28: error: 'mqttTopicParam' was not declared in this scope; did you mean 'mqttTopic'?
  162 |   iotWebConf.addParameter(&mqttTopicParam);
      |                            ^~~~~~~~~~~~~~
      |                            mqttTopic
C:/Users/nocturno/Documents/PlatformIO/Projects/220721-131454-nodemcuv2/src/esp8266_mrdiy_mqtt_local_notifier.ino:164:31: error: cannot convert 'boolean (*)()' {aka 'bool (*)()'} to 'std::function<bool(iotwebconf::WebRequestWrapper*)>'
  164 |   iotWebConf.setFormValidator(&formValidator);
      |                               ^~~~~~~~~~~~~~
      |                               |
      |                               boolean (*)() {aka bool (*)()}
In file included from C:/Users/nocturno/Documents/PlatformIO/Projects/220721-131454-nodemcuv2/src/esp8266_mrdiy_mqtt_local_notifier.ino:91:
.pio\libdeps\nodemcuv2\IotWebConf\src/IotWebConf.h:301:83: note:   initializing argument 1 of 'void iotwebconf::IotWebConf::setFormValidator(std::function<bool(iotwebconf::WebRequestWrapper*)>)'
  301 |   void setFormValidator(std::function<bool(WebRequestWrapper* webRequestWrapper)> func);
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
C:/Users/nocturno/Documents/PlatformIO/Projects/220721-131454-nodemcuv2/src/esp8266_mrdiy_mqtt_local_notifier.ino: In function 'boolean formValidator()':
C:/Users/nocturno/Documents/PlatformIO/Projects/220721-131454-nodemcuv2/src/esp8266_mrdiy_mqtt_local_notifier.ino:438:22: error: 'mqttServerParam' was not declared in this scope; did you mean 'mqttServer'?
  438 |   int l = server.arg(mqttServerParam.getId()).length();
      |                      ^~~~~~~~~~~~~~~
      |                      mqttServer
*** [.pio\build\nodemcuv2\src\esp8266_mrdiy_mqtt_local_notifier.ino.cpp.o] Error 1
============================================================================================================================= [FAILED] Took 19.93 seconds =============================================================================================================================
 *  El proceso del terminal "C:\Users\nocturno\.platformio\penv\Scripts\platformio.exe 'run'" finalizó con el código de salida 1. 
 *  Las tareas reutilizarán el terminal, presione cualquier tecla para cerrarlo.

RickeyWard, 2023-10-08:
I have a similar story, it this project doesn't want to build in platformio for esp32. Likely because nothing is versioned it just pointes to the latest everything
[image]

schmurtzm, 2023-10-09:
I haven't worked on this project since a while, that's true that the libraries should have versionned. The last commit is Aug 10, 2022, it was compiling correctly at this time so may be just versionning each library to this corresponding date should help.