schmurtzm/MrDiy-Audio-Notifier #23: Wifi AP not connecting

Bajkot, 2022-09-18:
Can see MrDIYNotifier-145574, but can't connect (wifi wrong password)
What do i do wrong?

Bajkot, 2022-09-18:
23:33:57.168 -> Config size: 357
23:33:57.168 -> Wrong config version. Applying defaults.
23:33:57.168 -> [iwcAll]
23:33:57.168 -> |-- [iwcSys]
23:33:57.168 -> |   |-- 'iwcThingName' with value: 'MrDIYNotifier-145574'
23:33:57.215 -> |   |-- 'iwcApPassword' with value: <hidden>
23:33:57.215 -> |   |-- [iwcWifi0]
23:33:57.215 -> |   |   |-- 'iwcWifiSsid' with value: ''
23:33:57.215 -> |   |   \-- 'iwcWifiPassword' with value: <hidden>
23:33:57.215 -> |   \-- 'iwcApTimeout' with value: '30'
23:33:57.215 -> |-- [iwcCustom]
23:33:57.215 -> |   |-- [mqttgroup]
23:33:57.215 -> |   |   |-- 'mqttServer' with value: ''
23:33:57.215 -> |   |   |-- 'mqttUser' with value: ''
23:33:57.215 -> |   |   |-- 'mqttPass' with value: <hidden>
23:33:57.215 -> |   |   \-- 'mqttTopic (without / at the end)' with value: ''
23:33:57.215 -> |   \-- [soundgroup]
23:33:57.215 -> |       |-- 'Sound_output' with value: ''
23:33:57.215 -> |       |-- 'Sound_Volume' with value: '0.5'
23:33:57.215 -> |       |-- 'samVoice :' with value: ''
23:33:57.262 -> |       \-- 'GoogleTTSvoice' with value: ''
23:33:57.262 -> \-- [hidden]
23:33:57.262 -> 
23:33:57.262 -> Using No DAC output on ESP8266, audio output on pins 
23:33:57.262 ->  speaker L & R -> GPIO3 (RX)
23:33:57.262 -> Don't try and drive the speaker pins can't give enough current to drive even a headphone well and you may end up damaging your device
23:33:57.262 -> SkipApStartup is requested, but either no WiFi was set up, or configButton was pressed.
23:33:57.262 -> AP password was not set in configuration
23:33:57.262 -> State changing from: 0 to 1
23:33:57.262 -> Setting up AP: MrDIYNotifier-145574
23:33:57.309 -> With default password: <hidden>
23:33:58.152 -> AP IP address: 192.168.4.1
23:33:58.152 -> AP password was not set.
23:33:58.152 -> WiFi SSID was not set.
23:33:58.152 -> Will stay in AP mode.
23:33:58.152 -> State changed from: 0 to 1
23:33:59.559 -> [MQTT] /status	idle

Bajkot, 2022-09-18:
And a strange row at the top
[image]

eokgnah, 2023-05-19:
how long is your wifi password?

with this i fixed my connection issues:
https://github.com/prampec/IotWebConf/issues/273