schmurtzm/MrDiy-Audio-Notifier #24: Tumping sound just before and after playing - ESP8266 no DAC

etlweather, 2022-09-22:
Works well! But when I play a sound or tone, it starts with a tumping sound, then plays and ends with two tumping sounds. Seems like this occurs when power is put on the pins.

This is with an ESP8266 with L/R wired to RX and GND to GND, plugged to a powered speaker.

etlweather, 2022-09-22:
Might be related to https://github.com/earlephilhower/ESP8266Audio/issues/406

etlweather, 2022-09-23:
Even more relevant: https://gitlab.com/MrDIYca/mrdiy-audio-notifier/-/issues/4

Hyperionlink, 2025-05-13:
So I've tried this setup with both esp8266 and esp32. The solution I found was to hook up an external DAC. It seems that the power supply isn't completely isolated and introduces quite a bit of noise. Adding a little bit of hardware in the form of the PCM5102 resolved this issue for me on both controllers.